compuphase/Black-Magic-Probe-Book #2: Building with GCC on Windows 10

sidprice, 2019-10-04:
HI, first thanks for all the information in the book.

I would like to build the tools with the GCC compiler but the makefiles seems targeted specifically to the Watcom compiler. Do you think I should create a separate makefile for my application or try to up[date the current file to work.

compuphase, 2019-10-05:
I have added Makefile.mingw for MingW. However, the makefile does not solve all problems. Apparently, MingW does not include strlcpy() and strlcat() in its libraries, so I added a C file with an implementation of those too. I could not find an include files for winusb either. Copying those from the Microsoft Windows DDK worked, but I don;t think I can redistribute those files. For required files are: usb.h, usb100.h, usb200.h, winusb.h, winusbio.h.

sidprice, 2019-10-05:
I would expect there to be a redist package for winusb. I don't have the DDK so I am not sure what is the best way forward.

sidprice, 2019-10-05:
@compuphase seems the file makefile.cfg is not in the repository.

compuphase, 2019-10-06:
I modified the makefiles such that makefile.cfg is optional. I also documented it at the top of the makefile. The CDIR macro is now optional too.

Including a makefile.cfg is our trick to allow for building on workstations that have tools and libraries at different paths. These workstation-specific locations are then in the makefile.cfg and the Makefile is universal. That is the reason why we don't put makefile.cfg in version control. (The alternative is, of course, to make sure all workstations have the same configuration, but that is not really easy on Microsoft Windows, when you wish to support multiple versions).

You already have the "redist" for winusb: the DLL is pre-installed. What you are missing are the header files. I do not recall installing the DDK on my machine, but I do have Visual Studio installed. Perhaps the DDK came with it.

sidprice, 2019-10-06:
I appreciate all the effort you are making here this is getting closer. Now the issue is that my installation does not have "makedepend". 

I also must ask why you did not use the GCC toolchain and just have a single makefile? Detecting the operating system is possible for selecting the appropriate differences and it is a much cleaner solution.

sidprice, 2019-10-06:
I did try a build and got a number of warnings plus several undefined symbols. Output from the linker is in this gist:

[Error Gist](https://gist.github.com/sidprice/61a5d9262a7398ade6a7131255678b97)

compuphase, 2019-10-06:
An implementation of makedepend is here: https://github.com/compuphase/makedepend

For the error, the following must be added to the linker lines: -lsetupapi -lshlwapi -lcomdlg32
I am stunned by the undefined reference to `__imp_GetStockObject'; that is just a plain Windows GDI function. Hopefully also adding -lgdi32 helps.

We use multiple C/C++ compilers. Some projects need to be done in Visual Studio. When we have a free choice, it is not uncommon that we choose two compilers. We find more bugs with this approach.
We have used Opus Make almost forever for our cross-platform universal Make. But it has not been updated since 1998, so I am moving away from Opus Make in favour of GNU Make. I have not switched everything over yet though.

sidprice, 2019-10-06:
Please see my [pull request](https://github.com/compuphase/Black-Magic-Probe-Book/pull/3)

Zagitta, 2019-10-14:
@sidprice What did you install you install and which folders did you add to the include path?
Trying to compile on win10 myself with mingw gcc and win10 WDK (formerly DDK) while setting the following in `makefile.cfg`:
`WINUSB_INC := -I"C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\um" -I"C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\shared"`
Just causes a bunch of compile errors seemingly related to undefined architecture and other random stuff.

sidprice, 2019-10-14:
@Zagitta I just did "make -f Makefile.mingw clean" followed by "make -f Makefile.mingw". THere were some warnings but the suite built.

compuphase, 2019-10-15:
@Zagitta I just updated the repository with (hopefully) improved support for MingW. I myself have struggled the most with WinUSB. So I added a winusb.def file to the project, that you can use to build the import library libwinusb.a with dlltool (from MingW).

Zagitta, 2019-10-24:
Thanks @compuphase, in the end I ended up just using the msvc nmake file instead where everything just worked out of the box.

closed 2023-02-24 by compuphase