compuphase/Black-Magic-Probe-Book #4: bmscan fails if multiple BMPs have been connected

sidprice, 2019-10-06:
I have more than one BMP, plus my own ctxLink probe which is based upon BMP.

The find_bmp method appears to only search for "interface" "0", here is a snapshot of the registry on my PC.

[image]

sidprice, 2019-10-06:
Sorry, I misunderstood the "interface", I need to dig deeper to find why the BMP is not being found.

sidprice, 2019-10-06:
I believe the issue is that the code is assuming there is only a single entry under "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\USB\VID_1D50&PID_6018&MI_00", when in fact on my system, because I have multiple BMPs and ctxLinks that have been attached I have five devices. The first device entry tested is not the one currently connected and so the find fails.

sidprice, 2019-10-06:
I think I have fixed this issue with this [PR](https://github.com/compuphase/Black-Magic-Probe-Book/pull/5)

compuphase, 2019-10-15:
Commit d228b6c31ede07c3049671f317bb15d6dc7f3210 fixes this issue (tested under Microsoft Windows and Linux).

closed 2019-10-15 by compuphase