compuphase/Black-Magic-Probe-Book #10: BMTrace - Trace access denied

sidprice, 2020-01-03:
I am trying to get BMTrace working with BMPon Windwos 10. My target is STM32F415RG. I have a logic analyzer on the SWO output from the MCU and I do see data being emitted. However, when I start BMTrace it reports "Trace Access Denied".

Suggestions please?

compuphase, 2020-01-04:
Looking in the source, I see two reasons why "trace access denied" is returned: CreateFile() failed to open the "device" for the trace interface, or WinUSB subsequently failed to initialize the interface (using the handle that CreateFile() returned).

I have just committed b2be091c8b2d3f4212bf0b4295d2dacaca59cfc7 which adds the value of GetLastError() to the message. Hopefully this will shed some light on the problem.

sidprice, 2020-01-04:
Thanks for the suggestions. In fact, I just ran BMTrace under a debugger and it appears that WinUsb_Initialize is failing. Turns out I had not installed the driver for the Trace Port :{

sidprice, 2020-01-04:
My next problem is that with BMTrace set to passive listener, I see output on the SWO pin, but BMTrace says just "Listening ..." no data is displayed

compuphase, 2020-01-04:
Hello Sid,
Most likely cause: the command "monitor traceswo" to the BMP is not given
in passive mode. I will look into it tomorrow. Passive mode is misdesigned.
One trick: say "monitor traceswo" in GDB once, then leave GDB and start
bmtrace. The BMP keeps traceswo on as long as it remains powered.

Regards,
Thiadmer

compuphase, 2020-01-07:
I just committed ca5d2199408647f0ae439a1e367c62d9e88ff147. This removes "passive mode" and adds two checkboxes: "configure target" and "configure debug probe". When both are unchecked, it is the old passive mode. With the new method, you can configure SWO tracing from your embedded code, but still give the "monitor traceswo" command (for the BMP) via the bmtrace utility. Also, there is more detailed error reporting.

sidprice, 2020-02-07:
Closing this issue.

closed 2020-02-07 by sidprice