compuphase/Black-Magic-Probe-Book #19: Trace Access Denied - error 122

sidprice, 2020-02-17:
I cannot get BMTrace to work (Windows), I see the above error message.

compuphase, 2020-02-18:
According to winerror.h, code 122 is ERROR_INSUFFICIENT_BUFFER (a data buffer passed to a system call is too small). This might be caused by structure padding.
Has BMTrace worked before, or has it never worked? If it did work before, any idea what changed?

sidprice, 2020-02-18:
Yes, BMTrace has previously worked, I don't remember any changes, perhaps my target is sending larger amounts of data, perhaps 64 bytes each USB packet

compuphase, 2020-02-20:
In commit b37136ed5d3626d4fee467d4a519f895a5f02797 I added an extra code to the error message, to hopefully pinpoint the problem.

sidprice, 2020-02-21:
Unfortunately the mingw build fails, the file "usb-support.h" appears to be missing.

compuphase, 2020-02-21:
Oops, I forgot to add the files to the project before commit. Fixed in commit bbc60b4c72ac6e36c70f329779beec441c4abfd1.

sidprice, 2020-02-21:
[image]

Error when running BMTrace.

compuphase, 2020-02-21:
Hmmm... so now the "access denied" error is magically gone and another error pops up. It feels to me like a memory corruption bug.
You can comment out this particular assertion; it does no harm. However, the situation simply should not occur because it implies that time jumps backward...

sidprice, 2020-02-22:
I commented out the assertion and BMTrace now works

compuphase, 2020-07-21:
Commit ffc80c9d9ca124d696b43da388e0fed89f2aeab6 has a fix for the bug signalled by the assertion. When multiple trace messages arrive in a single USB packet, the timestamp on the 2nd (and further) was set incorrectly.

closed 2020-07-21 by compuphase