compuphase/Black-Magic-Probe-Book #20: A suggestion

sidprice, 2020-02-17:
In the book the code for setting up the STM32 SWO trace always multiplies the input bitrate by 2. This means that the calling function must always supply 1/2 the required bitrate. It would be less confusing if the input bitrate were not multiplied by 2 I think.

Do you think this is a good idea?

compuphase, 2020-02-18:
There is the bit rate and the clock frequency. For Manchester encoding, there is an edge (rising or falling) halfway the bit period and possibly an edge at the start of the bit, That is why the clock runs at twice the frequency as the bit rate. So for a bit rate of 100 kbps, the clock must be set to 200 kHz.

When using asynchronous mode, the clock frequency must be equal to the bit rate. The book has no snippets for that yet. But in the BMTrace software, if you select asynchronous mode and let BMTrace configure the target, the frequency is not doubled. It is only doubled for Manchester.

sidprice, 2020-02-18:
What is confusing is that the init function in the book has the option to select async operation. Perhaps the Manchester/Asyn selection should be a parameter to that function and then the input bitrate is either doubled or not according to that selection. Less confusing I think :)

compuphase, 2020-07-21:
I have updated the code snippet and the description of it.

closed 2020-07-21 by compuphase