compuphase/Black-Magic-Probe-Book #22: Gtk:ERROR Fatal error reading PNG image file: IHDR: CRC error

koendv, 2020-05-09:
bmtrace crashes trying to save. 

```
Gtk-WARNING **: 16:25:35.259: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
apt-get remove fonts-freefont-ttf

Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/PiXflat/16x16/mimetypes/image.png: Fatal error reading PNG image file: IHDR: CRC error (gdk-pixbuf-error-quark, 0)
```
The cause seems to be that bmtrace etc. contain a function called "crc32". So does gtk. Changing the name of bmtrace's crc32() function solves the problem.
```
diff -urN ./ORIG/Black-Magic-Probe-Book/source/bmp-support.c ./Black-Magic-Probe-Book/source/bmp-support.c
--- ./ORIG/Black-Magic-Probe-Book/source/bmp-support.c	2020-05-09 16:44:08.857672975 +0200
+++ ./Black-Magic-Probe-Book/source/bmp-support.c	2020-05-09 16:43:42.458080717 +0200
@@ -532,7 +532,7 @@
     }
     fseek(fp, offset, SEEK_SET);
     fread(data, 1, filesize, fp);
-    crc_src = (unsigned)crc32((uint32_t)~0, data, filesize);
+    crc_src = (unsigned)bm_crc32((uint32_t)~0, data, filesize);
     free(data);
     /* request CRC from Black Magic Probe */
     sprintf(cmd, "qCRC:%lx,%lx",paddr,filesize);
diff -urN ./ORIG/Black-Magic-Probe-Book/source/crc32.c ./Black-Magic-Probe-Book/source/crc32.c
--- ./ORIG/Black-Magic-Probe-Book/source/crc32.c	2020-05-09 16:44:08.857672975 +0200
+++ ./Black-Magic-Probe-Book/source/crc32.c	2020-05-09 16:43:16.238486871 +0200
@@ -83,7 +83,7 @@
  *        first call, the value of the crc param is set to ~0, on every next
  *        call, it is set the the output of the previous call.
  */
-uint32_t crc32(uint32_t crc, const unsigned char *data, unsigned size)
+uint32_t bm_crc32(uint32_t crc, const unsigned char *data, unsigned size)
 {
   while (size--)
     crc = (crc << 8) ^ crc_table[((crc >> 24) ^ *data++) & 0xff];
diff -urN ./ORIG/Black-Magic-Probe-Book/source/crc32.h ./Black-Magic-Probe-Book/source/crc32.h
--- ./ORIG/Black-Magic-Probe-Book/source/crc32.h	2020-05-09 16:44:08.857672975 +0200
+++ ./Black-Magic-Probe-Book/source/crc32.h	2020-05-09 16:42:51.688868289 +0200
@@ -3,5 +3,5 @@
 #if defined __cplusplus
   extern "C"
 #endif
-uint32_t crc32(uint32_t crc, const unsigned char *data, unsigned size);
+uint32_t bm_crc32(uint32_t crc, const unsigned char *data, unsigned size);
 
```

FYI: There's another issue related to bmtrace here: https://github.com/Immediate-Mode-UI/Nuklear/issues/130#issue-615184515

compuphase, 2020-05-16:
Fixed in commit 074f1b953a952ba759aca2691e2ac175c1757cb5. Thank you.

closed 2020-05-16 by compuphase

koendv, 2020-05-16:
OK. Thanks.

BTW: https://github.com/blacksphere/blackmagic/wiki/Serial-Wire-Output

Pls. comment. My Windows knowledge is small.

koen

compuphase, 2020-05-19:
Hello Koen,

I still have to carefully read the wiki page that you linked. But I want to
say that it is a fantastic idea to decode SWO tracing on the BMP itself.
For many projects, it will make specialized trace viewers redundant.

When I get the time, I will test it and get back to you. For Windows 8 and
10, no drivers need to be installed for serial ports. For earlier versions,
one can use an ".inf" file that references the pre-installed serial driver.
The only other Windows-specific thing that I want to check is latency. In
the days of Windows XP, the pre-installed serial port driver buffered
incoming data until it reached a certain size or until no data was received
for 10ms. I have no idea whether that is still the case, or whether the
drivers have improved in the last 20 years.

Great work.

Regards,
Thiadmer Riemersma

koendv, 2020-05-20:
OK. I'm glad you like the idea. Also, feedback from Windows is welcome.
I've added your comment about the .inf file to the wiki.
There's one more feature which I'd like to implement: that, when decoding, a write to stimulus channel 31 flushes usb output buffers. I attach a patch that implements this feature, and a binary for "native". 
[blackmagic-flush-native.zip](https://github.com/compuphase/Black-Magic-Probe-Book/files/4656130/blackmagic-flush-native.zip)

regards,
koen

koendv, 2020-05-24:
Almost forgot: there's an Arduino library for SWO: https://github.com/koendv/SerialWireOutput