compuphase/Black-Magic-Probe-Book #29: bmdebug seg faults on Ubuntu 18.04

simonsunnyboy, 2023-02-08:
The bmdebug utility builds here and the GUI is briefly shown on screen but the application terminates without output in the console.

In the system log there is a notice on a segmentation fault:

`[xxxx.yyyyyy] bmdebug[18655]: segfault at 7fc97a1c39d0 ip 00007fc9880d4bd8 sp 00007fff62215200 error 4 in libpthread-2.27.so[7fc9880cc000+1a000]
`

compuphase, 2023-02-09:
I am unable to reproduce it (on Ubuntu 20.04), but I have a hunch. Can you try:
```
./bmdebug -g<path-to-arm-none-eabi-gdb>
```
One of the first steps bmdebug takes, is to pop up a file-open dialog to let you set the path to GDB.

simonsunnyboy, 2023-02-09:
That did the trick. 
User expectation would be to note the incompatible gdb selected by default on the console as a fault indicator.

compuphase, 2023-02-09:
The idea is that if there is no suitable GDB selected yet, bmdebug shows a popup that allows you to select it. However, there is apparently an issue with the selection dialog. The browse buttons ([...] at the right of some edit fields) also show a file selection dialog. Presumably, those still crash on your machine. So there is still a bug that I need to fix.

d-adamson, 2023-02-13:
Everything seems to be working using the BMP from the CLI.

However,I'm having the same or a similar issue with bmdebug. 
Running Linux Mint 21.1 (based on Ubuntu 22.04).
Happens both when 
1) using the dialog at startup and then navigating to navigating to /usr/bin/gdb-multiarch, and 
2) when specifying from the command line as follows
$ ./bmdebug -g/usr/bin/gdb-multiarch

free(): double free detected in tcache 2
Aborted (core dumped)

Note: The other utilities seem to be working, but I've only tested bmscan, bmserial and bmflash

**From /var/log/syslog**
systemd-coredump[21357]: Process 21327 (bmdebug) of user 1000 dumped core.#012#012Found module linux-vdso.so.1 with build-id: 6a61cf67fbc559c10e60ed0adeb9ab5e42248a3c#012Found module libdrm.so.2 with build-id: 9c7cb19295d20e515902cb0710326a0b8d6394c8#012Found module libxcb-glx.so.0 with build-id:

...

0bb435fdd5ec37178e14ea03bb36f779a4b72a94#012Found module bmdebug with build-id: 17bde16621a5feb35af1cf7aeba2cf7719d7d935#012Stack trace of thread 21327:#012#0  0x00007f2fc3919a7c __pthread_kill_implementation (libc.so.6 + 0x96a7c)#012#1  0x00007f2fc38c5476 __GI_raise (libc.so.6 + 0x42476)#012#2  0x00007f2fc38ab7f3 __GI_abort (libc.so.6 + 0x287f3)#012#3  0x00007f2fc390c6f
Feb 13 14:50:51 TOWER-PC-LINUX systemd[1]: systemd-coredump@9-21356-0.service: Deactivated successfully.

compuphase, 2023-02-14:
Probably an unrelated bug. Does this happen right after start-up (before you do anything else, such as set a breakpoint or start running)?

d-adamson, 2023-02-14:
It crashes immediately after selecting a file in the dialog.
Also crashes immediately when using the command line -g option.

By the way, the book is really great. It helped me a lot. In my opinion, it should be part of the official BMP documentation.

compuphase, 2023-02-15:
In commit 8ef5a140f99f0c2bbc826073f91ad0a8c5d7cc31, I added a "-d" switch for bmdebug, which logs the state that the program is in (in the terminal). Can you please try `bmdebug -d` and send me the result?

I have so far not been able to reproduce the issue. The "malloc debug library" that I use, does not find a double free. However, the error may be somewhere else. There is a lot happening after bmdebug starts GDB: scan for BMP port(s), checking firmware capabilities, loading the DWARF symbols... So with the extra information, I can hopefully pinpoint where to look.

Thank you for the compliment for the book. 1BitSquared is aware of it, but are unhappy with it. So it is an independent project.

d-adamson, 2023-02-15:
Hopefully 1BitSquared will eventually see the benefit of the book. 

This may be more info than you need, but trying to be thorough: 

**Here is my BMP in /dev**
ttyBmpGdb -> ttyACM0

**Here is the location of gdb-multiarch**
$ which gdb-multiarch 
/usr/bin/gdb-multiarch

****Command to launch and initial output:
 $ ./bmdebug -d**** 
State: 0 (moved from -1)
State: 1 (moved from -1)
State: 0 (moved from -1)

**Initial screenshot: It seems strange that gdb-multiarch is pre-populated in the File field.**
[image]

**Navigated to my ELF file:**
[image]

**Selected the GDB /usr/bin/gdb-multiarch**
[image]

**Program immediately crashes after selecting the GDB file: Here is the full output from the command line**
$./bmdebug -d
State: 0 (moved from -1)
State: 1 (moved from -1)
State: 0 (moved from -1)
State: 1 (moved from -1)
State: 2 (moved from -1)
State: 3 (moved from -1)
bmdebug: bmdebug.c:628: gdbmi_sethandled: Assertion `stringlist_getlast(&consolestring_root, STRFLG_RESULT, STRFLG_HANDLED) == NULL' failed.
Aborted (core dumped)

-------------------------------------------------------------------------------------------
**By the way, here is info from the build process:**

**my make.cfg file:**
GLFW_INC := /usr/include/GLFW
GLFW_LIB := -L/lib/x86_64-linux-gnu
GLFW_LIBNAME := glfw

**Here is the output from the build:**
`gcc -c /usr/include/GLFW -Wall -g -obmdebug.o bmdebug.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -oarmdisasm.o armdisasm.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -obmcommon.o bmcommon.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -obmp-scan.o bmp-scan.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -obmp-script.o bmp-script.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -odemangle.o demangle.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -odwarf.o dwarf.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -oelf.o elf.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -oguidriver.o guidriver.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -omcu-info.o mcu-info.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -omemdump.o memdump.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -ominIni.o minIni.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
[... 52 lines cut ...]
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -c /usr/include/GLFW -Wall -g -otcl.o tcl.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -L/lib/x86_64-linux-gnu -g -obmflash bmflash.o bmcommon.o bmp-scan.o bmp-script.o bmp-support.o cksum.o crc32.o elf.o gdb-rsp.o guidriver.o ident.o minIni.o nuklear_guide.o nuklear_mousepointer.o nuklear_style.o nuklear_tooltip.o rs232.o specialfolder.o tcl.o tcpip.o xmltractor.o findfont.o lodepng.o nuklear.o nuklear_glfw_gl2.o noc_file_dialog.o -lfontconfig -lglfw -lGL -lm -lbsd -ldl -lpthread -lX11 -lxcb -lXau -lXdmcp `pkg-config --libs gtk+-3.0` -lusb-1.0
gcc -c /usr/include/GLFW -Wall -g -obmprofile.o bmprofile.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -L/lib/x86_64-linux-gnu -g -obmprofile bmprofile.o bmcommon.o bmp-scan.o bmp-script.o bmp-support.o crc32.o decodectf.o demangle.o dwarf.o elf.o gdb-rsp.o guidriver.o mcu-info.o minIni.o parsetsdl.o rs232.o specialfolder.o swotrace.o tcpip.o xmltractor.o nuklear_guide.o nuklear_mousepointer.o nuklear_splitter.o nuklear_style.o nuklear_tooltip.o findfont.o lodepng.o nuklear.o nuklear_glfw_gl2.o noc_file_dialog.o -lfontconfig -lglfw -lGL -lm -lbsd -ldl -lpthread -lX11 -lxcb -lXau -lXdmcp `pkg-config --libs gtk+-3.0` -lusb-1.0
gcc -c /usr/include/GLFW -Wall -g -obmscan.o bmscan.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -L/lib/x86_64-linux-gnu -g -obmscan bmscan.o bmp-scan.o bmp-script.o bmp-support.o crc32.o elf.o gdb-rsp.o rs232.o specialfolder.o tcpip.o xmltractor.o -lbsd -lpthread
gcc -c /usr/include/GLFW -Wall -g -obmserial.o bmserial.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -L/lib/x86_64-linux-gnu -g -obmserial bmserial.o guidriver.o minIni.o rs232.o specialfolder.o noc_file_dialog.o nuklear_guide.o nuklear_mousepointer.o nuklear_splitter.o nuklear_style.o nuklear_tooltip.o tcl.o findfont.o lodepng.o nuklear.o nuklear_glfw_gl2.o -lfontconfig -lglfw -lGL -lm -lbsd -ldl -lpthread -lX11 -lxcb -lXau -lXdmcp `pkg-config --libs gtk+-3.0` -lusb-1.0
gcc -c /usr/include/GLFW -Wall -g -obmtrace.o bmtrace.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
[... cut ...]
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -L/lib/x86_64-linux-gnu -g -ocalltree calltree.o -lbsd
gcc -c /usr/include/GLFW -Wall -g -oelf-postlink.o elf-postlink.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -L/lib/x86_64-linux-gnu -g -oelf-postlink elf-postlink.o elf.o -lbsd
gcc -c /usr/include/GLFW -Wall -g -otracegen.o tracegen.c
gcc: warning: /usr/include/GLFW: linker input file unused because linking not done
gcc -L/lib/x86_64-linux-gnu -g -otracegen tracegen.o parsetsdl.o -lbsd`

simonsunnyboy, 2023-02-15:
> By the way, the book is really great. It helped me a lot. In my opinion, it should be part of the official BMP documentation.

I totally agree. Awesome read and cool to provide custom tailored tools. 

I surely hope 1BitSquared change their mind. It is promotion of their product afterall.

compuphase, 2023-02-16:
From the trace, it looks like bmdebug already chokes on the welcome message that GDB prints on start-up. What is the version of gdb-multiarch that you are using?

d-adamson, 2023-02-16:
Yes, It crashes as soon as I select the debug file.

$ gdb-multiarch -v
**GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1**
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

simonsunnyboy, 2023-02-16:
> What is the version of gdb-multiarch that you are using?

In case it is relevant, here is mine that I have to pass  on the command line. However it works fine here:

```
$ gdb-multiarch -v
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
```

compuphase, 2023-02-17:
@d-adamson: can you do another test with the latest version of bmdebug.c (just committed). There is more verbose output around the early stages (when running "./bmdebug -d").

It's curious: I booted Mint 21.1 from an USB stick, installed gdb-multiarch from the repository (which is version 12.1), and yet I still cannot reproduce the problem.

There is one change in the latest bmdebug (not directly related to the bug, but it came up because of this bug report): it now also searched for gdb-multiarch in the path (previous versions looked only for arm-none-eabi-gdb). So it should no longer be needed to add the path to GDB on the command line.

d-adamson, 2023-02-17:
Interesting. I started bmdebug -d but forgot to plug in my adapter and the program did not crash immediately. Even though it was pre-selected, I sucessfully selected /usr/bin/gbd-multiarch

Next, I plugged in my BMP adapter and relaunched bmdebug -d => crashed after a brief initialization phase.

I wonder if it could be related to the sym links used to give the probe a consistent name?
lrwxrwxrwx 1 root root 7 Feb 17 10:11 /dev/**ttyBmpGdb -> ttyACM0**
lrwxrwxrwx 1 root root 7 Feb 17 10:11 /dev/**ttyBmpTarg -> ttyACM1**

$ ./bmdebug -d

[ProgOutput.txt](https://github.com/compuphase/Black-Magic-Probe-Book/files/10768249/ProgOutput.txt)

compuphase, 2023-02-18:
The reason was the trailing space before the \n on the last line, plus that this line does not have a suffix with the architecture (M0+). Fixed in commit 98f08d40a1dbc560fc305c6505ce19fecf9847a9.

The other issue this reveals is: bmdebug does not offer you a choice which target to attach to. It will always attach to the first target in the list.

d-adamson, 2023-02-18:
Great! Sounds like it's getting close.

Sounds like a reasonable workaround - passing the "attach target" on the command line.

Unfortunately, I was not able to test the latest version. I got an error during the build.
Here is my **makefile.cfg** contents:
GLFW_INC := /usr/include/GLFW
GLFW_LIB := -L/lib/x86_64-linux-gnu
GLFW_LIBNAME := glfw

 **$make**
gcc -c /usr/include/GLFW -Wall -g -oguidriver.o guidriver.c
guidriver.c:552:5: error: conflicting types for ‘guidriver_appsize’; have ‘int(int *, int *)’
  552 | int guidriver_appsize(int *width, int *height)
      |     ^~~~~~~~~~~~~~~~~
In file included from guidriver.c:33:
guidriver.h:53:7: note: previous declaration of ‘guidriver_appsize’ with type ‘_Bool(int *, int *)’
   53 | bool  guidriver_appsize(int *width, int *height);
      |       ^~~~~~~~~~~~~~~~~
guidriver.c:574:5: error: conflicting types for ‘guidriver_poll’; have ‘int(int)’
  574 | int guidriver_poll(int waitidle)
      |     ^~~~~~~~~~~~~~
In file included from guidriver.c:33:
guidriver.h:55:7: note: previous declaration of ‘guidriver_poll’ with type ‘_Bool(_Bool)’
   55 | bool  guidriver_poll(bool waitidle);
      |       ^~~~~~~~~~~~~~
make: *** [Makefile:68: guidriver.o] Error 1

compuphase, 2023-02-19:
Sorry about that, fixed in commit 7b6f97414712e39ac5d898a6a4aa468bed1309d3.

d-adamson, 2023-02-19:
Almost there ...

bmdebug comes up and initializes, but I do not see anything in the source window.
However, I can type "list main" in the console and see the source.
Normally my *.elf is in a "build" subdirectory. I've tried putting the *.elf and *.c in the same directory, but that does not seem to help.

[image]

d-adamson, 2023-02-19:
Also, here is the content of the blink-10.elf.bmcfg file:

[Target]
entrypoint=main
cmsis-svd=
[Settings]
tpwr=0
connect_srst=0
[Flash]
auto-download=0
[SWO trace]
mode=1
bitrate=100000
clock=48000000
datasize=8
enabled=0
ctf=
chan0="0 #000000 "
chan1="0 #000000 "
chan2="0 #000000 "
chan3="0 #000000 "
chan4="0 #000000 "
chan5="0 #000000 "
chan6="0 #000000 "
chan7="0 #000000 "
chan8="0 #000000 "
chan9="0 #000000 "
chan10="0 #000000 "
chan11="0 #000000 "
chan12="0 #000000 "
chan13="0 #000000 "
chan14="0 #000000 "
chan15="0 #000000 "
chan16="0 #000000 "
chan17="0 #000000 "
chan18="0 #000000 "
chan19="0 #000000 "
chan20="0 #000000 "
chan21="0 #000000 "
chan22="0 #000000 "
chan23="0 #000000 "
chan24="0 #000000 "
chan25="0 #000000 "
chan26="0 #000000 "
chan27="0 #000000 "
chan28="0 #000000 "
chan29="0 #000000 "
chan30="0 #000000 "
chan31="0 #000000 "
[Serial monitor]
mode=0
port=
baud=0

d-adamson, 2023-02-19:
Here are some files for reference:

blink-10.c
blink-10.elf
blink-10.bmcfg
blink-10_cli_output.txt  (output from the command line running with ./bmdebug -d )

[blink-10.zip](https://github.com/compuphase/Black-Magic-Probe-Book/files/10778001/blink-10.zip)

d-adamson, 2023-02-19:
Also, I can step through the source when running gdb-multiarch from the command line with tui -enable

[image]

compuphase, 2023-02-23:
This one left me scratching my head for a while. The debug log shows that GDB gives the full paths to the source files, and GDB can read those source files itself, so why would bmdebug fail to read them?

Then I noticed that some source files appear multiple times in the list provided by GDB. For GDB, the list of files is a table, and it refers to the "current file" by a zero-based index into that table. The bmdebug front-end stores the GDB table as an array, but... it does not load the same source file twice. So that is what happened: the GDB "current file index" no longer aligns with the bmdebug array index.

I have redesigned how bmdebug handles source files and references to source files. It no longer uses an array, but explicitly stores the GDB source table indices with each source file (and there can be multiple GDB indices for a single source file).

This is in commit 747334bd986d93edc68fa989d694edd3c1eb9570. The caveat is that I have so far only done limited testing on it myself.

compuphase, 2023-02-28:
These issues all appear fixed as of commit 56cd68372dfd06fb0acaa555f2070bef32337810.