compuphase/Black-Magic-Probe-Book #31: bmflash: add support for STM32 option bytes

simonsunnyboy, 2023-02-15:
bmflash at this moment only supports removal of option bytes for LPC devices.
Erasing/resetting of STM32 devices would be handy. 

At the moment the user still has to use other programmers and tools for this.

compuphase, 2023-02-16:
The option "Erase Option Bytes" under the "Tools" button is for STM32 microcontrollers. The LPC devices do not use option bytes, but instead a 4-byte magic value at a fixed address in Flash ROM.

simonsunnyboy, 2023-02-16:
In this case I suggest to label the option accordingly.
I will test again as I didn't get any feedback when applying this.

compuphase, 2023-03-15:
The way to set Flash read protection has changed in commit ccc75fe7427b48f7f3130718914d19ee78e7c10b. You can now choose level 1 or 2 for read protection (note that level 2 is irriversible). Level 1 can be reverted with "Erase Option Bytes" under the "Tools" button.