compuphase/Black-Magic-Probe-Book #34: Source Code Not Visible

d-adamson, 2023-02-23:
Moving this to a new issue

bmdebug comes up and initializes, but I do not see anything in the source window.
However, I can type "list main" in the console and see the source.
Normally my *.elf is in a "build" subdirectory. I've tried putting the *.elf and *.c in the same directory, but that does not seem to help.

[image]

_Originally posted by @d-adamson in https://github.com/compuphase/Black-Magic-Probe-Book/issues/29#issuecomment-1436095470_

d-adamson, 2023-02-23:
Just tested commit 747334b

Unfortunately the source is still not visible. Skimming through the changes, I can see you put some serious work into this.

Here is the latest screenshot, output file, and bmcfg file.

In the screenshot, the gdb console is pointing to the correct source file.

I can set break points in console and the code will run to the break point.

[image]
[Output_bmdebug.txt](https://github.com/compuphase/Black-Magic-Probe-Book/files/10815143/Output_bmdebug.txt)
[blink-10.elf.bmcfg.txt](https://github.com/compuphase/Black-Magic-Probe-Book/files/10815288/blink-10.elf.bmcfg.txt)

compuphase, 2023-02-24:
In commit 31cd6ab9e3c48a31bed5818d6e18cb0ca436a3a4, I made two changes:

1) Extra logs (when run with "bmdebug -d" option) to show what source file it attempts to load, and what the result was.

2) When the main view says "NO SOURCE", the line below shows GDB's index in its table with source files, and which filename matches that number according to bmdebug.

With the answers to these points, we can narrow down on the bug.

d-adamson, 2023-02-24:
Linux make failed
$ make
gcc -c /usr/include/GLFW -Wall -g -obmdebug.o bmdebug.c
bmdebug.c: In function ‘sources_add’:
bmdebug.c:1597:46: error: ‘errno’ undeclared (first use in this function)
 1597 |       printf("file open failed, error %d\n", errno);
      |                                              ^~~~~
bmdebug.c:100:1: note: ‘errno’ is defined in header ‘<errno.h>’; did you forget to ‘#include <errno.h>’?
   99 | # include "res/icon_debug_64.h"
  +++ |+#include <errno.h>
  100 | #endif
bmdebug.c:1597:46: note: each undeclared identifier is reported only once for each function it appears in
 1597 |       printf("file open failed, error %d\n", errno);

Added line 57 in bmdebug.c
# include <errno.h>

Make completed.

Starting testing ...

d-adamson, 2023-02-24:
Here are the results:
[image]
[Out.txt](https://github.com/compuphase/Black-Magic-Probe-Book/files/10828434/Out.txt)

compuphase, 2023-02-25:
Finally found the parsing error. Should be fixed in commit c5f03b1c9da82bd0a29c92dc5dc92157d6abf061.

d-adamson, 2023-02-25:
Fantastic! I just checked it and it's working! :+1: 

Unfortunately, there appears to be another crash when opening a SVD file.

path to svd
~/pico/pico-sdk/src/rp2040/hardware_regs/rp2040.svd

Here is the log output:

[OutSVD.txt](https://github.com/compuphase/Black-Magic-Probe-Book/files/10832026/OutSVD.txt)

compuphase, 2023-02-26:
What I see, in the OutSVD.txt file, is that the code started running with a temporary breakpoint at main, then moves to the "STOPPED" state when it reaches the breakpoint. At that point, the log is aborted. There was probably still data in the stdout buffer at the time of the crash.

In the commit c3442c223c0132b57e16b437979d058e5a5d6566, I only added fflush() to the log functions, so that the debug output will be complete.

If the program halted correctly at main() and that crashed as soon as you opened the SVD file, the bug is likely in the SVD parser. In that case, can you send me the rp2040.svd. I got a copy from the raspberrypi/pico-sdk project here on GitHub, but that loaded fine in my case. I tested on Windows, though. I will test on Linux tomorrow. (But the code is generic, so I do not see how it would be OS-dependent.)

d-adamson, 2023-02-26:
Unfortunately, it is still crashing. I waited about 30 seconds after the code hit the break point at main, just to let any potential background tasks settle.

Here is the output of the log file and the svd file (I added the .txt extension so that it would upload to this thread).

[OutSvd-2.txt](https://github.com/compuphase/Black-Magic-Probe-Book/files/10834613/OutSvd-2.txt)
[rp2040.svd.txt](https://github.com/compuphase/Black-Magic-Probe-Book/files/10834617/rp2040.svd.txt)

compuphase, 2023-02-27:
Commit ca36470a56978170eb001b7b298ec56127e80ec0 fixes a buffer overrun in SVD parsing.

I was able to reproduce the crash on a different workstation. After this fix, loading the SVD file no longer crashed.

d-adamson, 2023-02-27:
Thanks Thiadmer ,

It looks like it is all up and running! 
You are a super hero! :1st_place_medal:

compuphase, 2023-02-27:
Thanks a lot for your persistence in testing. I would not have found these three (!) bugs without your log data.

closed 2023-02-27 by compuphase