compuphase/Black-Magic-Probe-Book #35: bmdebug Reset stuck at Initializing. Please wait ...

d-adamson, 2023-03-12:
Hi,
Pressing "Reset" does not always reset. Prog seems to get stuck at "Initializing. Please wait..."

Sequence of events that leads to the issue.
after program load 
Cont -> Stop -> Reset -> program gets stuck and have to restart it

See attached screenshot and debug log.

[image]
[bmdebug.log](https://github.com/compuphase/Black-Magic-Probe-Book/files/10952263/bmdebug.log)

compuphase, 2023-03-15:
What I see in the log, is that there is no response to the "stop". I see that the command is given: the next-to-last line in the log is "State: 21". This is the result on clicking on the continue/stop button (and since the firmware was running, it must be "stop"). However, there is no response from GDB.

The last line, "State: 4", is the response to clicking on "reset". Again, there is no response from GDB.

A work-around it: pressing Ctrl+C twice within 3 seconds does a full reset.

It is puzzling why GDB does not respond to the "stop" command. Are you able to stop the program in (bare) GDB?

d-adamson, 2023-03-15:
While my program is running, I'm not able to execute any commands (no gdb prompt). Same in bmdebug and gdb-multiarch. 

When stuck initializing, pressing Ctrl+C multiple times does not stop the program.

After unplugging and replugging the BMP. It seems to be working again. However from time-to-time, it gets stuck again. Should have some time toward the end of the week to see if I can find a repeatable pattern.

d-adamson, 2023-03-17:
Found a pattern that may be impacting this... or it could be a separate issue - not sure.

When I do a reset, half of the time the program returns to the breakpoint at main(). The other half, the program continues running without stopping at main(). 

**This behavior is not random; it alternates:**
reset -> stops at main()
reset -> continues running w/o stopping
reset -> stops at main()
reset -> continues running w/o stopping
...

Each time the program continues running without stopping after a reset, there is an error message:
error,msg="Error erasing flash with vFlashErase packet"

Looking at the debug log, this error just occurs after a "compare-sections" command
When the line shows "4040 compare-sections" => no error message follows => restarts at main()
When the line shows "c040 compare-sections" => error message follows => restart does not work as expected

Hope this is helpful.
[bmdebug.log](https://github.com/compuphase/Black-Magic-Probe-Book/files/11004755/bmdebug.log)

d-adamson, 2023-03-17:
Another clue.
The above only seems to happen when the "Download to target on mismatch" option is selected.

d-adamson, 2023-03-22:
Here are two log files.
One when working correctly and one when failing. These were run back-to-back with no modifications to either my target app nor to bmdebug. 

Recall: this only seems to happen when "Download to target on mismatch" is selected.

I'll take a shot at tracking it down, but this one may be above my pay level. :)
[bmdebug-working.txt](https://github.com/compuphase/Black-Magic-Probe-Book/files/11041068/bmdebug-working.txt)
[bmdebug-failing.txt](https://github.com/compuphase/Black-Magic-Probe-Book/files/11041069/bmdebug-failing.txt)

compuphase, 2023-03-22:
I have yet to look more deeply into this, but I can confirm a few of your findings:

1) The "reset" button indeed does exactly the same as the "reset" command.

2) Given the error message "Error erasing flash with vFlashErase packet", this is definitely linked to the "Download to target on mismatch" option. When that option is selected, bmdebug first issues a "compare-sections" command to check whether the code in Flash in the microcontroller matches the ELF file. On mismatch, it then proceeds to download the ELF file. The first step in downloading to Flash is to erase all Flash sectors that will be rewritten. That procedure fails.

When the option "Download to target on mismatch" is _not_ set, bmdebug skips the downloading. It still runs the "compare-sections" command, to give a warning on mismatch.

When you give a "reset load" command, bmdebug skips the "compare-sections" command, but proceeds immediately with downloading the ELF file to the microcontroller.