compuphase/Black-Magic-Probe-Book #36: Command line history not available on Linux

d-adamson, 2023-03-19:
Hi again,

Started using "reset load" as a work around to the initialization issue (#35).
That's when I discovered that the command history is not working on Linux (ctrl+up arrow, ctrl+r ...).

d-adamson, 2023-03-20:
Got it working!

When running bmdeug on a new machine, there is no command history and newly added commands are not added to the history stack.

Added the following two lines to the end of the file ~/.config/BlackMagic/bmdebug.ini and it works now:
```
[Commands]
cmd1=next
```

I'll propose a fix in the source code in the coming days.

compuphase, 2023-10-04:
Fixed by commit 216f595f8c604c2d00987005bf151dfeef6dd9a0 (pull request #37).

closed 2023-10-04 by compuphase