compuphase/Black-Magic-Probe-Book #41: BMDebug: DWARF error. Wrong version in compilation unit header (is 5, should be 2,3 or 4)

OevreFlataeker, 2024-01-20:
No sure, if that's the proper place to report this.
I am using bmdebug.exe from this repo's releases and try to follow some examples with the Black-Magic-Probe book.
When I try to load a Blink.ino.elf from Arduino IDE 2 with enabled debug symbold, bmdebug crashes after showing this message in the message pane:

BMDebug: DWARF error. Wrong version in compilation unit header (is 5, should be 2,3 or 4) (in module <path to elf>)

Is this something I need to change in the Arduino IDE or is this a matter against which version BMDebug.exe has been compiled against maybe?

OevreFlataeker, 2024-01-20:
[Blink2.ino.elf.zip](https://github.com/compuphase/Black-Magic-Probe-Book/files/13998712/Blink2.ino.elf.zip)
Example Blink2.ino.elf.zip as a reference

compuphase, 2024-01-22:
This message appears to come from GDB (BMDebug is a front-end for GDB, so error messages from GDB are shown in the console view).
Check that BMDebug is using the correct GDB. In the configuration section (top right of the BMDebug interface), you can set the path to the GDB to use. Generally speaking, you should use the GDB that comes with the GCC compiler that you used to build the firmware with. It looks like GCC is a fairly recent version, but that the GDB launched by BMDebug is quite old.

OevreFlataeker, 2024-01-22:
Thanks! Makes perfectly sense! Yes the ARM toolchain bundled with Arduino Studio is probably much newer than the GDB from the ARM toolchain I had on another path on my system! Will try!

closed 2024-10-10 by compuphase