compuphase/Black-Magic-Probe-Book #42: OrbTrace mini not recognized as a probe.

ohunter, 2024-01-29:
I think it'd be nice to be able to use the OrbTrace Mini along with these utilities but that doesn't seem to be possible. I suspect this has to do with the way that `find_bmp()` is looking for the device. I have not tested this on windows so it might be possible that it is detected there without any issues.

compuphase, 2024-01-29:
The utilities locate the probe on their VID:PID numbers. The OrbTrace Mini uses different numbers than the Black Magic Probe. So you are correct: `find_bmp()` does not locate the OrbTrace.

However, the utilities also use the GDB protocol and the `monitor` command strings to set up the debug probe. So it will only work if OrbTrace is (by and large) compatible with the Black Magic Probe.

I do not own an OrbTrace Mini, so I cannot check.

ohunter, 2024-01-30:
The orbtrace mini is a compatible hosted probe that supports blackmagic yes.

retif, 2024-06-05:
OrbTrace is CMSIS-DAP v1 and v2 compatible