compuphase/Black-Magic-Probe-Book #45: Crash: bmdebug: dwarf.c:1287: dwarf_linetable: Assertion `prologue.version<5' failed.

retif, 2024-06-05:
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)) 13.2.1 20231009
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
```
$ ./bmdebug ~/projects/orbmule/examples/simple/ofiles/simple.elf 
bmdebug: dwarf.c:1287: dwarf_linetable: Assertion `prologue.version<5' failed.
Aborted (core dumped)
```
```
$ readelf --debug-dump=info ~/projects/orbmule/examples/simple/ofiles/simple.elf 
Contents of the .debug_info section:

  Compilation Unit @ offset 0x0:
   Length:        0x335 (32-bit)
   Version:       5
   Unit Type:     DW_UT_compile (1)
   Abbrev Offset: 0x0
   Pointer Size:  4
 <0><c>: Abbrev Number: 10 (DW_TAG_compile_unit)
    <d>   DW_AT_producer    : (indirect string, offset: 0x1938): GNU C99 13.2.1 20231009 -mthumb -mcpu=cortex-m7 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -march=armv7e-m+fp -g3 -gdwarf-5 -ggdb -Og -std=gnu99 -ffunction-sections -fdata-sections -fmerge-constants -fno-common -ffreestanding -fno-builtin -fstack-usage
    <11>   DW_AT_language    : 12	(ANSI C99)
    <12>   DW_AT_name        : (indirect string, offset: 0x11cd): src/itm_messages.c
    <16>   DW_AT_comp_dir    : (indirect string, offset: 0x2b15): /home/oleks/projects/orbmule/examples/simple
    <1a>   DW_AT_ranges      : 0xc
    <1e>   DW_AT_low_pc      : 0x0
    <22>   DW_AT_stmt_list   : 0x0
    <26>   DW_AT_macros      : 0x0
 <1><2a>: Abbrev Number: 2 (DW_TAG_base_type)
    <2b>   DW_AT_byte_size   : 8
    <2c>   DW_AT_encoding    : 7	(unsigned)
    <2d>   DW_AT_name        : (indirect string, offset: 0x346e): long long unsigned int
 <1><31>: Abbrev Number: 2 (DW_TAG_base_type)
    <32>   DW_AT_byte_size   : 4
    <33>   DW_AT_encoding    : 7	(unsigned)
    <34>   DW_AT_name        : (indirect string, offset: 0x313d): unsigned int
 <1><38>: Abbrev Number: 2 (DW_TAG_base_type)
    <39>   DW_AT_byte_size   : 1
    <3a>   DW_AT_encoding    : 8	(unsigned char)
    <3b>   DW_AT_name        : (indirect string, offset: 0x2aea): unsigned char
 <1><3f>: Abbrev Number: 2 (DW_TAG_base_type)
    <40>   DW_AT_byte_size   : 2
    <41>   DW_AT_encoding    : 7	(unsigned)
    <42>   DW_AT_name        : (indirect string, offset: 0x474): short unsigned int
 <1><46>: Abbrev Number: 2 (DW_TAG_base_type)
    <47>   DW_AT_byte_size   : 1
    <48>   DW_AT_encoding    : 6	(signed char)
    <49>   DW_AT_name        : (indirect string, offset: 0x2966): signed char
 <1><4d>: Abbrev Number: 2 (DW_TAG_base_type)
    <4e>   DW_AT_byte_size   : 2
    <4f>   DW_AT_encoding    : 5	(signed)
    <50>   DW_AT_name        : (indirect string, offset: 0x61f): short int
 <1><54>: Abbrev Number: 2 (DW_TAG_base_type)
    <55>   DW_AT_byte_size   : 4
    <56>   DW_AT_encoding    : 5	(signed)
    <57>   DW_AT_name        : (indirect string, offset: 0x27c8): long int
 <1><5b>: Abbrev Number: 2 (DW_TAG_base_type)
    <5c>   DW_AT_byte_size   : 8
    <5d>   DW_AT_encoding    : 5	(signed)
    <5e>   DW_AT_name        : (indirect string, offset: 0x15fa): long long int
 <1><62>: Abbrev Number: 6 (DW_TAG_typedef)
    <63>   DW_AT_name        : (indirect string, offset: 0x1ae5): uint8_t
    <67>   DW_AT_decl_file   : 2
    <67>   DW_AT_decl_line   : 46
    <68>   DW_AT_decl_column : 24
    <69>   DW_AT_type        : <0x38>
 <1><6d>: Abbrev Number: 6 (DW_TAG_typedef)
[... 662 lines cut ...]
    <5c5>   DW_AT_external    : 0
    <5c6>   DW_AT_type        : <0x5f4>
    <5c8>   DW_AT_low_pc      : 0x8000219
    <5cc>   DW_AT_high_pc     : 2
 <1><5cd>: Abbrev Number: 2 (DW_TAG_subprogram)
    <5ce>   DW_AT_name        : (indirect string, offset: 0x74a6): SysTick_Handler
    <5d2>   DW_AT_external    : 0
    <5d3>   DW_AT_type        : <0x5f4>
    <5d5>   DW_AT_low_pc      : 0x8000219
    <5d9>   DW_AT_high_pc     : 2
 <1><5da>: Abbrev Number: 2 (DW_TAG_subprogram)
    <5db>   DW_AT_name        : (indirect string, offset: 0x74b6): Default_Handler
    <5df>   DW_AT_external    : 0
    <5e0>   DW_AT_type        : <0x5f4>
    <5e2>   DW_AT_low_pc      : 0x8000219
    <5e6>   DW_AT_high_pc     : 2
 <1><5e7>: Abbrev Number: 2 (DW_TAG_subprogram)
    <5e8>   DW_AT_name        : (indirect string, offset: 0x74c6): DEF_IRQHandler
    <5ec>   DW_AT_external    : 0
    <5ed>   DW_AT_type        : <0x5f4>
    <5ef>   DW_AT_low_pc      : 0x8000219
    <5f3>   DW_AT_high_pc     : 2
 <1><5f4>: Abbrev Number: 3 (DW_TAG_unspecified_type)
 <1><5f5>: Abbrev Number: 0
```

compuphase, 2024-06-05:
Thanks, I need to update GCC and test. Up to version 12, the line table was still version 3.

compuphase, 2024-06-12:
Can you give me more information on the command line options for GCC?

I use GCC 13.2.1 (like you), and a build a sample program with this:
```
arm-none-eabi-gcc -gdwarf-5 -mcpu=cortex-m3 -mthumb -o blinky.elf blinky.c -nostartfiles -Wl,-T,STM32F103x8.ld
```

With objdump, I can confirm that the symbolic information is in DWARF version 5:
```
arm-none-eabi-objdump --dwarf=info blinky.elf
```
```
blinky.elf:     file format elf32-littlearm

Contents of the .debug_info section:

  Compilation Unit @ offset 0:
   Length:        0x9da (32-bit)
   Version:       5
   Unit Type:     DW_UT_compile (1)
   Abbrev Offset: 0
   Pointer Size:  4
 <0><c>: Abbrev Number: 21 (DW_TAG_compile_unit)
    <d>   DW_AT_producer    : (indirect string, offset: 0x5b): GNU C++17 13.2.1 20231009 -mcpu=cortex-m3 -mthumb -mfloat-abi=soft -march=armv7-m -gdwarf-5
    <11>   DW_AT_language    : 12	(ANSI C99)
    <12>   DW_AT_name        : (indirect string, offset: 0x366): blinky.c
    <16>   DW_AT_comp_dir    : (indirect string, offset: 0x147): d:\Tools\BlackMagicProbe\blinky\stm32bluepill
    <1a>   DW_AT_low_pc      : 0x8000150
    <1e>   DW_AT_high_pc     : 0xb0
    <22>   DW_AT_stmt_list   : 0
 <1><26>: Abbrev Number: 5 (DW_TAG_base_type)
    <27>   DW_AT_byte_size   : 1
-- More  --
```

However, the line number table (`.debug_line`) still uses version 3 in my case:
```
arm-none-eabi-objdump --dwarf=rawline blinky.elf
```

```
blinky.elf:     file format elf32-littlearm

Raw dump of debug contents of section .debug_line:

  Offset:                      0
  Length:                      719
  DWARF Version:               3
  Prologue Length:             606
  Minimum Instruction Length:  2
  Initial value of 'is_stmt':  1
  Line Base:                   -5
  Line Range:                  14
  Opcode Base:                 13

 Opcodes:
  Opcode 1 has 0 args
  Opcode 2 has 1 arg
  Opcode 3 has 1 arg
  Opcode 4 has 1 arg
  Opcode 5 has 1 arg
-- More  --
```

I have tried compiling in C++ mode, but with the same result (`.debug_line` table still in version 3).
So there appears to be a compiler option that causes GCC to emit the `.debug_line` table in version 5.
I am running these tests on Windows, by the way.

compuphase, 2024-06-12:
I find a way to create a version 5 `.debug_line` table: add the option `-Wa,-gdwarf-5`.
It turns out that the `.debug_line` table is often created by the assembler.

compuphase, 2024-06-13:
Support for DWARF version 5 is added in commit 67feb1954b3f9b8586104c6fc6f75db6ade88161.

closed 2024-06-13 by compuphase